WayneKeenan/picraftzero #47: forward and backward reversed when rock candy connected directly to macos

thymjan, 2017-06-20:
I connected my rock candy controller directly to my mac. The control is as smooth as directly connected to the raspberry. And it's plug and play!

How can I inverse backward and forward of the rock candy controller connected directly to Safari/macos?

Found this in the picraftzero journalctl:
`Joystick(id=0, xname=rx, yname=ry, xinvert=False, yinvert=False` 
So I think I have to insert yinvert=True in the config file for the right joystick id.
Am I right?

WayneKeenan, 2017-06-20:
The config that installed along with the PiConZero python library and used as the library defaults can be found on GitHub here [picraftzero/resources/config/picraftzero.cfg](https://github.com/WayneKeenan/picraftzero/blob/master/picraftzero/resources/config/picraftzero.cfg)

I'm planning to keep putting all possible config sections and keys there (and do the doc asap).
Please use that config file as a reference and copy the 'joystick_0' section from that to /etc/picraftzero.cfg

At the moment the joystick config applies to both the web and 'local' joypad.  Is this ok for you or do you need something more fine grained?

thymjan, 2017-06-20:
Ok, thanks. I put the following section to my config file:
```
[joystick_0]
invert_x_axis=no
invert_y_axis=yes
```
Now - as you said - the direction of the rock candy joypad connected to the browser is correct but the on screen joystick appears inversed.

WayneKeenan, 2017-06-20:
I'll look at this a bit more.

WayneKeenan, 2017-06-20:
I've made a fix (but not a release yet) which will make things more consistent. It will also mean you to undo the axis inversion in /etc/piconzero.cfg

thymjan, 2017-06-20:
Hey, that's nice and quick! Sorry for the question, but can you give me a hint where the bubbleworks/input.js file is located in the raspbian-tree?

thymjan, 2017-06-20:
... found it myself:
find / -name inputs.js -print
/usr/local/lib/python3.4/dist-packages/picraftzero/resources/www/js/lib/bubbleworks/inputs.js

:-)

WayneKeenan, 2017-06-20:
If you wanted, you could uninstall and install straight from the git repo and ride the bleeding edge:

```bash
sudo pip3 uninstall picraftzero
sudo pip3 install git+https://github.com/WayneKeenan/picraftzero
```

When picraftzero starts there is a line in the log messages that shows the location of the config files, one of which is where the picraftzero package is installed:

```bash
2017-06-20 20:38:02,189 -        config.py: 20 -             parse_config() ] - INFO - Config file search locations: ['/usr/local/lib/python3.4/dist-packages/picraftzero/resources/config/picraftzero.cfg', '/etc/picraftzero.cfg', '/home/pi/.picraftzero.cfg']
```
Based on the above you will find `` here:
```bash
/usr/local/lib/python3.4/dist-packages/picraftzero/resources/www/js/lib/bubbleworks/inputs.js
```

WayneKeenan, 2017-06-20:
Please let me know if the joypad fix works for you and I'll close the issue.

thymjan, 2017-06-20:
Perfect! Steering reacts as expected now. Thank you!

WayneKeenan, 2017-06-20:
cool, thank you.

closed 2017-06-20 by WayneKeenan

thymjan, 2017-06-20:
... but with installing straight from the repo I lost my live-stream. Only black window now.

WayneKeenan, 2017-06-21:
Please post the log messages

thymjan, 2017-06-21:
I have to reboot twice. First call of the IP-adress was extremely slow, than it worked again. I think this was the same at my first install. Unfortunately I didn't observe the log. Next time it happens I will try to investigate.