WayneKeenan/picraftzero #48: Where's a place to do indiviudal things

thymjan, 2017-06-20:


WayneKeenan, 2017-06-20:
It may help for me to mention you can add 'PiCraftZero features' to any kind of standalone python script, e.g. gpiozero scripts.    For example the [CamJam robot with PiCraftZero controls added](https://github.com/WayneKeenan/picraftzero/blob/master/examples/camjam_edukit3.py), can be run as 
```python
python3 camjam_edukit3.py
```

However, at the moment PiCraftZero just deals with joysticks, sorry, I've not yet added button presses and I've not documented how to use the HTML/JS/WebSocket parts of PiCraftZero in order for someone to be able to create their own UI components and send custom messages (but it's doable and part of 'the plan').

I am up for taking requests so please raise an issue/request for this and anything else you think needs adding.  I think adding couple of buttons/switches to the default UI and a 'button press' would be a nice thing to have.

I'd also like to mention that if you've enabled the system service you can also start your own script, instead of the default PiCraftZero script, by adding a line to a 'service' section in the PiCraftZero config.  

Here's some steps todo that, e.g. if you have a script at `/home/pi/my_robot.py`

### Running your own script as a service

Edit the PicraftZero config file:

```bash 
sudo nano /etc/picraftzero.cfg
```

Add these lines: 
```bash
[service]
script=/home/pi/my_robot.py
```

Follow the log file, in the background, before you restart the service: 
```bash 
sudo journalctl -f -u picraftzero & 
```

Restart the service: 
```bash
sudo systemctl restart picraftzero
```

You should see in the logs something like:

` ... Running user script: /home/pi/my_picraftzero.py `

Your script will now run instead of the default PiCraftZero script every time the Pi boots.

thymjan, 2017-06-20:
Maybe this is a bit stupid (I'm not a professional coder...) but I tried to run the following python-code parallel to picraftzero:
```
#!/usr/bin/env python
# coding: Latin-1
# Load library functions we want
from explorerhat import output

output.one.on()
output.two.on()

try:
    print('Press CTRL+C to quit')

    while True:
       x = 0

except KeyboardInterrupt:
    output.one.off()
    output.two.off()

print("bye")
```

The LEDs flicker (normally they should stay on till exit). And by exiting I can't move the rover with picraftzero anymore. Seems to reset the explorer pHat. So this may not be trivial. Also the movement of the Tiny4wd is not as smooth than without the script.

thymjan, 2017-06-20:
By the way: what's that LED-setting in the [camera]-section about?

WayneKeenan, 2017-06-20:
When running your own script standalone you need to not run `python3 -m picraftzero` and/or stop the picraftzero service with:
```bash
sudo systemctl stop picraftzero
```

WayneKeenan, 2017-06-20:
the LED in the camera section is to enable/disable the operation of the red LED indicating when the camera is capturing.

WayneKeenan, 2017-06-20:
I made some modifications to the `explorer` library and bundle the it picraftzero, for now.

please try replacing:
```python
from explorerhat import output
```

with
```python
from picraftzero.thirdparty.pimoroni.explorerhat import output
```

thymjan, 2017-06-20:
I made the modifications and started the code with:
```
python3 -m LightsOn.py 
2017-06-20 23:13:46,703 -          zero.py: 18 -                 <module>() ] - INFO - picraftzero, version=0.2.3
2017-06-20 23:13:46,717 -          zero.py: 32 -                 <module>() ] - INFO - Selected pin factory is <class 'gpiozero.pins.rpigpio.RPiGPIOPin'>
2017-06-20 23:13:54,414 -      joystick.py: 44 -                 <module>() ] - INFO - Optional BlueDot library not found
2017-06-20 23:13:54,419 -      joystick.py: 73 -                 <module>() ] - INFO - Using EventDev implementation
2017-06-20 23:13:54,545 -      __init__.py:724 -                 <module>() ] - INFO - Explorer pHAT detected...
Press CTRL+C to quit
^Cbye
/usr/bin/python3: Error while finding spec for 'LightsOn.py' (<class 'AttributeError'>: 'module' object has no attribute '__path__')
2017-06-20 23:15:01,549 -      __init__.py:685 -         explorerhat_exit() ] - INFO - 
Explorer HAT exiting cleanly, please wait...
```
LEDs are still flickering, control is still much slower, no change of behavior. After CTRL+C picraftzero-gui doesn't work anymore.

WayneKeenan, 2017-06-21:
I've not replicated your setup but that 'while True' loop you have is a 'busy loop' and is going to be a CPU hog, explains why things have become less responsive.   
There is a more efficient way to be 'idle' in python using the `signal` module and `pause()`, which is what the `start()` call uses under the hood in the script below.

Please try this:
```python

#!/usr/bin/env python
# coding: Latin-1
# Load library functions we want

from picraftzero import Joystick, Wheelbase, steering_mixer, start
from picraftzero.thirdparty.pimoroni.explorerhat import output 

joystick = Joystick() 
motors  = Wheelbase(left=0, right=1)
motors.source = steering_mixer(joystick.values) 

output.one.on()
output.two.on()

try:
    print('Press CTRL+C to quit')

    start()

except KeyboardInterrupt:
    output.one.off()
    output.two.off()

print("bye")
```

WayneKeenan, 2017-06-21:
For got to add that you start the script by running it 'as normal' without the -m flag:
```bash
python3  LightsOn.py 
```

WayneKeenan, 2017-06-21:
raised a explorer hat question with Pimoroni here: https://github.com/pimoroni/explorer-hat/issues/34

WayneKeenan, 2017-06-21:
Adding an issue to record the need to upstream some of the changes made in the bundled Explorer library https://github.com/WayneKeenan/picraftzero/issues/50

thymjan, 2017-06-21:
The modified [explorer_lights.py](https://github.com/WayneKeenan/picraftzero/blob/master/examples/explorer_lights.py) works excellent with the LEDs now, no flickering anymore.

thymjan, 2017-06-21:
The rock candy joypad has several unused buttons. Is it possible to react on a button press?

Within the pimoroni [tinypirate example python script](https://github.com/Coretec-Robotics/Tiny_4wd/blob/master/TinyPirate.py) there are possibilities to do so:
```
 if event.code == "BTN_TL":
                if event.state == True:
                    print("Botton Left")
```

WayneKeenan, 2017-06-22:
Cool, good to see it's working. I will raise a separate issue for adding button support.

closed 2017-06-22 by WayneKeenan

WayneKeenan, 2017-06-22:
please see #53