WayneKeenan/picraftzero #52: After using joypad the on screen control isn't smoothly anymore

thymjan, 2017-06-21:
with macos/safari

WayneKeenan, 2017-06-22:
I'm not able to replicate this using the default picraftzero script.
It would help if you could provide a bit more detail:

1. What script are you running?
2. Which version of picraftzero are you using?  
3. Is picraftzero installed on a fresh SDCard image without anything else being setup or running?
4. When this occurs what is consuming CPU resource, e.g what is at the top of the output of `top`?

Thanks.

WayneKeenan, 2017-08-13:
Closing due to lack of activity. Please reopen with the requested info.

thymjan, 2017-08-13:
Sorry for the missing answers. Here they are:
1. does not depend
2. actual version
3. yes
4. python3 with increases from about 40% to 50% CPU

I can reproduce it either in Safari and Firefox on macOS.
This are the steps:
1. The rock candy receiver is attached to the mac
2. call the PiCraft website
3. Use the on screen control with the mouse. Works smoothly.
4. Use the rock candy controller. Runs smoothly either.
5. Try to take control with the mouse again. Hiccup.
6. Keyboard Keys are working fine.
7. Rock candy controller is still working fine.
8. Reload the PiCraft website by pressing the Camera Symbol in the footing.
9. Mouse control works smoothly again.

Hope this helps.