WayneKeenan/picraftzero #53: Support buttons

WayneKeenan, 2017-06-22:
Support Joypad buttons.
Add buttons to the web UI

Probably using the same API as gpio zero but instead of pins use logical id.  Also, use text or constant logical id's, e.g 'BTN_X' or BTN_X instead of numeric ones used for joysticks

WayneKeenan, 2017-06-22:
Added an example [here](https://github.com/WayneKeenan/picraftzero/blob/master/examples/buttons.py)

thymjan, 2017-06-22:
Should this example script work or is it still a draft? I can't get it working with my rock candy joypad. I connected the controller directly to the raspi and to safari. Unfortunately no success.

WayneKeenan, 2017-06-23:
Do you get an error?
Please send the log messages.

Did you also update picraftzero library from GitHub?

thymjan, 2017-06-23:
I don't have to edit the example file (button definitions), do I?
I updated from git. I don't receive error messages. By button press nothing happened. I started the script with "python3 button.py" pointed the focus to the web-page and pressed the buttons. Nothing happened in the log file.

WayneKeenan, 2017-06-23:
You don't have to edit the example, it should work as-is, but only when the USB dongle is plugged into the Pi.   I should clarify that it wont send button presses via the web client, yet. (When I tested on Mac I was running the `button.py` script on my Mac, not a Pi, to check the pygame backend)

I've pushed an update that has a bit more debug:
```bash
sudo pip3 uninstall picraftzero
sudo pip3 install git+https://github.com/WayneKeenan/picraftzero
```

Please enable debug : 

`sudo nano /etc/picraftzero.cfg `

Add:
```
[logging]
debug_enabled=yes
```

Please re-run this on a Pi with the RockCandy dongle attached to the Pi and post the log output, thanks.

Hopefully you will see something such as:

```log
ssh://pi@tiny4wd.local:22/usr/bin/python3 -u /home/pi/pycharm.projects/PiCraft/examples/buttons.py
2017-06-23 07:55:56,647 -      joystick.py: 47 -                 <module>() ] - INFO - Optional BlueDot library not found
2017-06-23 07:55:56,650 -      joystick.py: 76 -                 <module>() ] - INFO - Using EventDev implementation
2017-06-23 07:55:56,655 -          zero.py: 19 -                 <module>() ] - INFO - picraftzero, version=0.2.3
2017-06-23 07:55:56,662 -          zero.py: 33 -                 <module>() ] - INFO - Selected pin factory is <class 'gpiozero.pins.rpigpio.RPiGPIOPin'>
2017-06-23 07:55:57,420 -      joystick.py:120 -                 __init__() ] - INFO - Found input devices: ['/dev/input/event0']
2017-06-23 07:55:57,422 -      joystick.py:123 -                 __init__() ] - INFO - Using device: /dev/input/event0
2017-06-23 07:55:57,428 -      joystick.py:130 -                 __init__() ] - INFO - Device USB VPID: 3695:296
2017-06-23 07:55:57,430 -        config.py: 20 -             parse_config() ] - INFO - Config file search locations: ['/home/pi/pycharm.projects/PiCraft/picraftzero/resources/config/picraftzero.cfg', '/etc/picraftzero.cfg', '/home/pi/.picraftzero.cfg']
2017-06-23 07:55:57,462 -        config.py: 25 -             parse_config() ] - INFO - Merging config from these files:
2017-06-23 07:55:57,467 -        config.py: 27 -             parse_config() ] - INFO - ['/etc/picraftzero.cfg',
2017-06-23 07:55:57,469 -        config.py: 27 -             parse_config() ] - INFO -  '/home/pi/pycharm.projects/PiCraft/picraftzero/resources/config/picraftzero.cfg']
2017-06-23 07:55:57,471 -        config.py: 40 -               get_config() ] - INFO - Combined final config is:
2017-06-23 07:55:57,520 -        config.py: 42 -               get_config() ] - INFO - {'camera': {'framerate': '24',
2017-06-23 07:55:57,523 -        config.py: 42 -               get_config() ] - INFO -             'hflip': 'no',
2017-06-23 07:55:57,525 -        config.py: 42 -               get_config() ] - INFO -             'led': 'no',
2017-06-23 07:55:57,527 -        config.py: 42 -               get_config() ] - INFO -             'rotation': '0',
2017-06-23 07:55:57,529 -        config.py: 42 -               get_config() ] - INFO -             'vflip': 'yes'},
2017-06-23 07:55:57,531 -        config.py: 42 -               get_config() ] - INFO -  'hmd': {'camera_left_url': 'http://cam0.local:8080/stream/video.mjpeg',
2017-06-23 07:55:57,533 -        config.py: 42 -               get_config() ] - INFO -          'camera_mono_url': 'http://${WINDOW_LOCATION_HOSTNAME}:8080/stream/video.mjpeg',
2017-06-23 07:55:57,535 -        config.py: 42 -               get_config() ] - INFO -          'camera_right_url': 'http://cam1.local:8080/stream/video.mjpeg',
2017-06-23 07:55:57,537 -        config.py: 42 -               get_config() ] - INFO -          'camera_view_lx': '13',
2017-06-23 07:55:57,539 -        config.py: 42 -               get_config() ] - INFO -          'camera_view_ly': '40',
2017-06-23 07:55:57,541 -        config.py: 42 -               get_config() ] - INFO -          'camera_view_rx': '20',
2017-06-23 07:55:57,543 -        config.py: 42 -               get_config() ] - INFO -          'camera_view_ry': '60',
2017-06-23 07:55:57,545 -        config.py: 42 -               get_config() ] - INFO -          'iconbar_element': 'icon-bar'},
2017-06-23 07:55:57,548 -        config.py: 42 -               get_config() ] - INFO -  'inputs': {'keyboard_mapping': "'8BitDoZero'"},
2017-06-23 07:55:57,550 -        config.py: 42 -               get_config() ] - INFO -  'joystick_0': {'invert_x_axis': 'no', 'invert_y_axis': 'no'},
2017-06-23 07:55:57,552 -        config.py: 42 -               get_config() ] - INFO -  'joystick_1': {'invert_x_axis': 'no', 'invert_y_axis': 'no'},
2017-06-23 07:55:57,554 -        config.py: 42 -               get_config() ] - INFO -  'logging': {'debug_enabled': 'yes',
2017-06-23 07:55:57,556 -        config.py: 42 -               get_config() ] - INFO -              'web_debug_enabled': 'no',
2017-06-23 07:55:57,558 -        config.py: 42 -               get_config() ] - INFO -              'websocket_logging_enabled': 'no'},
2017-06-23 07:55:57,560 -        config.py: 42 -               get_config() ] - INFO -  'pantilt': {'pan_angle_offset': '0', 'tilt_angle_offset': '0'},
2017-06-23 07:55:57,563 -        config.py: 42 -               get_config() ] - INFO -  'services_topology': {'hosts_services_mapping': "{'tiny4wd.local': "
2017-06-23 07:55:57,565 -        config.py: 42 -               get_config() ] - INFO -                                                  "['picraftzero']}, "
[... cut ...]
2017-06-23 07:56:12,187 -      joystick.py:161 -                   _start() ] - DEBUG - event at 1498204572.186267, code 04, type 04, val 589827, event at 1498204572.186267, code 04, type 04, val 589827
2017-06-23 07:56:12,190 -      joystick.py:155 -                   _start() ] - DEBUG - KeyEvent, keycode=BTN_C, scancode=306, keystate=0
2017-06-23 07:56:12,196 -      joystick.py:161 -                   _start() ] - DEBUG - event at 1498204572.186267, code 00, type 00, val 00, synchronization event at 1498204572.186267, SYN_REPORT 
```

WayneKeenan, 2017-06-23:
I have noticed a bug where the very first button press doesn't call the `when_pressed` handler, will fix.

thymjan, 2017-06-24:
Got this working now when attaching the rock candy transceiver directly to the raspberry pi.
Here is the logging:
[button_log.txt](https://github.com/WayneKeenan/picraftzero/files/1100031/button_log.txt)

WayneKeenan, 2017-06-26:
great, thanks for confirming.