WayneKeenan/picraftzero #57: How to get output from user script

thymjan, 2017-08-13:
I reading analog and digital inputs from the Explorer pHAT in a user script.

How can I get the output?

Sorry, I'm a newbie to this but my `print('blabla')` doesn't reach the logfile with journalctl nor the output to command line.

The killer feature would be an overlay text to the Web-Video screen, would it?
So you can see if your battery state is ok and so on.

WayneKeenan, 2017-08-13:
try this, should work (but not actually tested)

```python
from picraftzero.log import logger

logger.info("my message")
```

The `uv4l` camera streaming service can have text overlaid,  and [seems very configurable](https://www.linux-projects.org/uv4l/tutorials/text-overlay/). 
The other way would be todo it in HTML/JS.

WayneKeenan, 2017-08-13:
If this doesn't work, please reopen.  There are a few other tickets for adding overlaid info.

closed 2017-08-13 by WayneKeenan

thymjan, 2017-08-13:
Thanks for your help. Meanwhile I exactly found this way to get the outputs to the log file:
```
from picraftzero.log import logger
...
logger.info(">>Switch lights off...")
```