WayneKeenan/picraftzero #58: error after upgrading to recent raspbian

thymjan, 2017-08-22:
the start procedure ran into following error:
Aug 22 09:52:57 zerow systemd[1]: picraftzero.service holdoff time over, scheduling restart.
Aug 22 09:52:57 zerow systemd[1]: Stopping PiCraftZero Service...
Aug 22 09:52:57 zerow systemd[1]: Starting PiCraftZero Service...
Aug 22 09:52:57 zerow systemd[1]: Started PiCraftZero Service.
 Aug 22 09:53:04 zerow python3[3860]: 2017-08-22 09:53:04,301 -      joystick.py: 47 -                 <module>() ] - INFO - Optional BlueDot library not found
Aug 22 09:53:04 zerow python3[3860]: 2017-08-22 09:53:04,313 -      joystick.py: 76 -                 <module>() ] - INFO - Using EventDev implementation
Aug 22 09:53:04 zerow python3[3860]: 2017-08-22 09:53:04,331 -          zero.py: 19 -                 <module>() ] - INFO - picraftzero, version=0.2.4 build=development
Aug 22 09:53:04 zerow python3[3860]: /usr/bin/python3: Error while finding spec for 'picraftzero.__main__' (<class 'AttributeError'>: 'module' object has no attribute 'pin_factory'); 'picraftzero' is a package and cannot be directly executed
Aug 22 09:53:04 zerow systemd[1]: picraftzero.service: main process exited, code=exited, status=1/FAILURE
Aug 22 09:53:04 zerow systemd[1]: Unit picraftzero.service entered failed state.

WayneKeenan, 2017-08-22:
By recent I guess you mean 'Stretch' ?   
It comes with a newer version of GPIOZero (1.4) which has changed it's API a bit and PiCraftZero needs to catch up.

thymjan, 2017-08-22:
Yes, actually it's based on debian stretch.
uname -r shows:
4.9.35+

WayneKeenan, 2017-10-26:
Some initial touch testing shows the latest release (0.2.12) works on Stretch.

closed 2017-10-26 by WayneKeenan