WayneKeenan/picraftzero #60: ios gamepad-controller: forward / backward reversed

thymjan, 2018-05-19:
I‘ve connected a steelseries wireless gamepad-controller to my iPad: After pressing the „A“-button it worked flawlessly!
Only the direction for forward / backward is reversed.
Is there an option to change the assignment for directions with ios-devices?

WayneKeenan, 2018-05-19:
There is a global and user config file, it's not specific to iOS clients though.

This is the template https://github.com/WayneKeenan/picraftzero/blob/master/picraftzero/resources/config/picraftzero.cfg

Look at the 'joystick_0' section

It can be placed in /etc/picraftzero.cfg

WayneKeenan, 2018-05-19:
Or ~/.picraftzero.cfg

WayneKeenan, 2018-05-19:
You only need to insert the section you want to override

thymjan, 2018-05-19:
Thanks for your help. Did as you told me. Now the Rock Candy controller with macos is inversed. The ios onscreen touch control also.

WayneKeenan, 2018-05-19:
As I mentioned, it's global, not iOS or device specific, that's the way it is at the moment.

thymjan, 2018-05-19:
Do you remember issue [https://github.com/WayneKeenan/picraftzero/issues/47](https://github.com/WayneKeenan/picraftzero/issues/47)? I mentioned the same problem with macos. You wrote to always invert joypad y-axis with html5. So macos and ios doesn't have the same html5 behaviour? So the bug is in macos?

WayneKeenan, 2018-05-20:
no, forgot all about that, what version are you using in regards to this issue?  PyPi release or installed from Git ?

thymjan, 2018-05-20:
I think I've did this:
_update to the latest repository_
```
sudo pip3 uninstall picraftzero
sudo pip3 install git+https://github.com/WayneKeenan/picraftzero
```
.. some time has passed since then.