WayneKeenan/picraftzero #64: HTML

rbooth2018, 2018-09-08:
How would I go about running Picraftzero through HTML,  so I could control my robot outside my network.

WayneKeenan, 2018-09-10:
Do you mean access you robot remotely via your home router when on the internet away from home?

rbooth2018, 2018-09-10:
Away from home... Do I have to port forward.  Pretty new to robotics and
rapsberry pie.   Also uses the noobs install of rapsbian, your universal
remote works great saved me hours of writing code.  However I have a slight
problem with my adafruit pan tilt mount.  I have the servos plugged in the
0 and 1 sockets. But after a few pan and tilt movements the servos freeze
and I have to restart picraft zero.  Once restarted it does it again...

Any help would be greatly appreciated

Thanks

Ritchie